PRBonn/OverlapNet #2: Questions about Evaluation Performance

kxhit, 2020-06-12:
Hi! Thanks for sharing your work! This work is quite interesting.
However, I have some confusion about the experiments.

1. How about the performance on KITTI 08? As we know, KITTI 08 has reverse loops, which is much more challenging than KITTI 00, 02, 05. I didn't see results on KITTI 08.
2. [Scan Context](https://github.com/irapkaist/scancontext) is a great loop closure detection method and achieve state-of-the-art performance on KITTI dataset. Are you going to report the results of comparision with Scan Context?
3.  As we know, it is important for algorithm to be robustness to view point changes (rotation in yaw) and occlusions. I didn't see the relevent evaluation in paper. 

Looking forward to your reply!

Best,
Xin

Chen-Xieyuanli, 2020-06-12:
Hey @kxhit,

I'm very happy to hear from you again!

1. Thank you for reminding me that KITTI 08 is more challenging. I'll later also try our method on KITTI 08.

2. Scan Context is a great method and Prof. Kim Ayoung is famous in the robotics domain. I didn't compare to their method since it's more like a map-based method while our method is a frame-based method. I'll also later try to use different datasets as they used like NCLT, robotcar, etc. 

3. The overlap is rotation invariant. More detail about how to calculate overlap you could now find in our paper at equation 3. Later, more details could found in the code.

Thank you again for following our work and contributing to our SuMa++ and rangenetlib before.
For more details, we could discuss via email or WeChat :-)

Best,
Yuanli

kxhit, 2020-06-12:
Hi again! @Chen-Xieyuanli 
Thanks for your prompt reply!
As I know, Scan Context is a frame-based method with single frame as input, and I contributed a little part of [implementation with Python](https://github.com/irapkaist/scancontext/pull/2). Welcome to discuss with me if I can offer help! 
Looking forward to the open-source code and further evaluation results! Thanks!

Happy coding!
Xin

Chen-Xieyuanli, 2020-06-26:
Hey @kxhit,

The first version of OverlapNet has been released. Please feel free to give a try.

I am going to close this issue. If there are any other questions, please feel free to drop me msgs.

Thank you for following our work.

Best,
Yuanli

closed 2020-06-26 by Chen-Xieyuanli