PRBonn/OverlapNet #4: Some questions.

kissb2, 2020-06-26:
Thank you for your very nice work. 
This is not an issue about the code, but I have general questions about the OverlapNet, and I couldn't wait for the RSS officially starts :)
in detail, the definition of the overlapness and the idea helped me.

The questions are:
1. If we do not have a semantic channel (that is a very general option so far), the performance degradation is expected or not, if there is, how much you expect?
2. Does the alg. supports the overlap estimation (and loop detection) between the (same sensor but) mounted at the different systems (e.g., robot 1 and robot 2) / or between the different sessions?
3. I'd like to know how much the system is, empirically, resistible to the existence of dynamic objects. I think the deep learning-based loop detection is good for the kind of object-aware or -robust things.

thank you!

Chen-Xieyuanli, 2020-06-29:
Hey @kissb2, 

Thank you a lot for following our work and for your great questions!

1. The semantic information is a kind of extra information from learning-based semantic segmentation which will definitely help the network to distinguish different places better. For a more detailed ablation study, you could find it in our paper Sec IV-E.

2. We didn't make any experiments like that so far. I guess it will still work to predict the overlap between two LiDAR scans. If the scenes are not too different, a loop closing/ICP still may work. If you use different robots, then the mounting height of the LIDAR is important: If the height is too different, then the scans may have no/low (ground-truth) overlap even when you are at the same position and the scene has not been changed at all. Here I assume you use a lidar with 32 or 64 beams that have a small vertical field of view. Therefore an ICP between such scans may also be difficult. If the height is the same (and the lidar is not tilted), it should work.

3. We test our method on KITTI and Ford campus datasets, which are collected from the real world with dynamic objects and it works well. We didn't explicitly design any architecture to remove dynamic objects while trained our light-weight network in an end-to-end fashion. I think the multiple cues from the LiDAR scan will help the network resist the dynamic objects. One could also explicitly use geometric and semantic information to get rid of the dynamic objects like what we did in our [SuMa++](https://github.com/PRBonn/semantic_suma) to improve the overall odometry and mapping results.

I hope this helps!

kissb2, 2020-07-06:
Thank you for the kind explanation!

closed 2020-07-06 by kissb2