PRBonn/OverlapNet #5: Performance on 16 beam lidars

zhangjun-xyz, 2020-08-05:
Hi Chen, I am really impressed on OverlapNet's excellent performance even using cheap geometric information of point clouds. One main concern of mine is how it works on sparser point clouds, e.g., produced by VLP-16. It's really common in industrial fields that you can't afford 64-beam lidars. 
Thank you for your great work and looking forward to your reply.

Chen-Xieyuanli, 2020-08-05:
Hey @zhangjun-xyz, 

thank you for following our work!

This is a very good question. We haven't tried other types of LiDAR sensors so far. However, since the network is quite light and simple, one could easily adapt OverlapNet to other types of input. The only part that needs to be adapted is the shape of the leg. 

We currently have no plan to test other types of sensors and warmly welcome you and other users to fork our repo to give a try. If you have any further questions please feel free to drop me msgs or send me e-mails. My e-mail address is: xieyuanli.chen@igg.uni-bonn.de

Best,

zhangjun-xyz, 2020-08-05:
I understand. It's unrealistic for you guys to test all the sensors on the market. I am just curious about the performance on vlp-16 and see whether you've done it. 
I am not sure whether I am able to test it myself. But if I do this I will let you know.

closed 2020-08-05 by zhangjun-xyz