PRBonn/OverlapNet #6: Question about the semantic_prob data

CK-Sung, 2020-08-07:
Hi I have a question about the semantic_prob data.

I would like to test with other scan data in KITTI, but I got an error message from the semantic probs that "cannot reshape array of size # into shape (20)". 

So I have question below.

1. Does the label data in the 'data/semantic_probs' are the RangeNet++ result?

2. What if the answer of the question 1 is YES, what is difference between the RangeNet++ label result and SemanticKITTI label data?

3. How can I save the RangeNet++ label data to adapt in the OverlapNet?

Thank you.

Chen-Xieyuanli, 2020-08-07:
Hey @CK-Sung,

1. The semantic_probs are the results of the second last layer of RangeNet++ which contains the probabilities over all the classes. 

2. As far as I know, the SemanticKITTI dataset only provides the ground truth labels but not the probabilities over all the classes. You could try directly training OverlapNet with label values. However, since the labels are discrete, the results may not very good. You could also generate "probabilities" like dataset for training OverlapNet using SemanticKITTI by assigning the real class channel as 1 and other classes as 0. 

3. As I answered [here](https://github.com/PRBonn/rangenet_lib/issues/31), you could use [rangenetlib](https://github.com/PRBonn/rangenet_lib) to generate the semantic estimates with probabilities over all classes.

I hope it helps and thank you for following our work!

CK-Sung, 2020-08-07:
Thank you for answer and great works!!

closed 2020-08-07 by CK-Sung