PRBonn/OverlapNet #7: Questions about training procedure

oet93, 2020-09-17:
Hi, thank you for your interesting work.

I am trying to train a model in order to achieve the results obtained in your paper on the KITTI odometry dataset. I followed the steps that you described in this repository and trained the model with multiple KITTI sequences by exploiting also intensity and semantic information. However, the performances I obtained are not good.
Therefore my questions are:

1. There are network parameters that need to be changed with respect to the default ones?

2. I noticed that in demo4, when ground truth are generated for the sequence 07, a data normalization step is performed in order to balance the data by considering the overlap rate. However, it seems that the example that you provided is calibrated on that run, therefore how can I perform this task when multiple sequences are considered?
For example, some KITTI sequences have only few samples with an overlap rate >0.5 (e.g. sequence 03 contains only one sample that matches this condition) and at the moment I perform such balancing by considering the overlap distribution across all the KITTI sequences.

3. Can you explain what "use_class_probabilities_pca" parameter is (network.yml)?

If you could give me any suggestion, it would be great! :) 

Thank you and great work!

Chen-Xieyuanli, 2020-09-18:
Hey @oet93, thank you for following our work!

1. The default parameters are that we used for the paper.

2. We used seq 03-10 for training. For each sequence, we did the normalization separately to keep a roughly balanced distribution of the overlap rate. You could also normalize the distribution over all KITTI sequences. We didn't try that but it should also work. There should be many samples with the overlap rate > 0.5 even in seq 03, since at least each frame will have a 100% overlap with themselves, and the scans nearby should also have a larger overlap.

3. It's a little bit tricky to use semantic information. The semantic information we used for this paper is not the semantic labels but the predicting probabilities over all classes from rangenet++.  Instead of using all probabilities of 20 classes, we did a PCA first and mapped the probabilities into three-channel which reduced the redundancy between similar classes and accelerate the performance. So here 'use_class_probabilities_pca' means we are using the semantic probabilities are PCA.

There are several tips which you may find useful are: 1. start to train the OverlapNet with only geometry information. Once you get good results and then add more different cues. 2. It's the same with multiple heads. It's better to start with overlap prediction only, and later then add other heads for other functionalities.

I hope this helps!

oet93, 2020-10-22:
Hi, thank you for your reply!

At the moment, I tried to achieve the performances of your pre-trained model that exploits only geometric information. I obtained some improvements by following your suggestions, however I cannot reach similar results.
I think that I'm making some mistakes when I generate the groundtruth for each sequence by using the demo4 script. Considering a sequence, do you simply compute the overlaps for each frame and then you perform the normalization of the overlaps distribution? If not, can you provide additional information of this step?

I have another question about the usage of semantic information. Once OverlapNet is trained by using only geometry information, how can I load the pre-trained model in order to accept additional channels as input?

Thanks again for your help.

Chen-Xieyuanli, 2020-10-25:
Hey @oet93,

Yes, I just compute the overlap and normalize the distribution afterward. 

Could you post somehow the results of the training or evaluation like other users did here: #9?
You could also have a look at the results in #9. The results there seem good.

Once OverlapNet is trained it is hard to add other information since the input shape is fixed.

Recently we are busy with our icra paper. I, therefore, cannot provide further help for your training.
But I will appreciate it a lot if you could provide more training results here, and I probably can later provide more help on this issue.

Chen-Xieyuanli, 2020-12-14:
Hey @oet93,

Is there any update on this issue?

Chen-Xieyuanli, 2021-01-06:
Since there is no further update about this issue, I'm going to close it.

If there is any other further question about this, please feel free to ask me to reopen it.

closed 2021-01-06 by Chen-Xieyuanli