PRBonn/OverlapNet #12: Some questions about training models

lilin-hitcrt, 2020-11-30:
Thanks for your amazing work. I have some questions about overlapnet. 
1. I found that demo4 only generates the overlaps between the first frame and other frames, but ground_truth_mapping[:,0] is set to len(scan_paths)-1.
2. Did you only use the overlaps between the first frame and other frames during training, or did you use other sampling methods?
3. There is a random selection in the training code, but this function is disabled by default in the configuration file. Does this have an impact on the result?
4. In the process of testing with the pre-trained model, we found that the random rotation of the data will have a significant impact on the results, but we found that you said that overlapnet is rotation invariance.

Chen-Xieyuanli, 2020-12-01:
Hey @lilin-hitcrt,

Thanks for following our work and contributing to our codes.

These are very good questions!

1. Yes, we provide demo4 just for a quick example to show how to calculate the ground truth overlaps. We therefore only calculate the overlaps between the last frame and all previous frames. During the training, we will use the overlaps between all frames with a normalized distribution.

2. See 1.

3. We actually didn't use the random selection in the training phase, since we find the training pairs are enough to train a good model using the OverlapNet without an augmentation. 
Besides the code used for the paper, this implementation also contains some other useful functions. One could also give them a try, and this random selection probably will lead to a more general model.

4. This is a very good question! Theoretically, the overlap estimation should be rotation-invariant, since the delta head used to estimate the overlap just computes the grid-wise difference between two features, and the rotation in yaw angle is just horizontal shifting in range images, which should not influence the results of the delta head.
However, when we train the two heads together sharing the same legs, this assumption may change due to the influence of the correlation head, since the correlation head is not rotation-variant.

lilin-hitcrt, 2020-12-02:
> Hey @lilin-hitcrt,
>
> Thanks for following our work and contributing to our codes.
>
> These are very good questions!
>
>     1. Yes, we provide demo4 just for a quick example to show how to calculate the [...]

Thanks for your quickly reply

lilin-hitcrt, 2020-12-02:
> Hey @lilin-hitcrt,
>
> Thanks for following our work and contributing to our codes.
>
> These are very good questions!
>
> 1. Yes, we provide demo4 just for a quick example to show how to calculate the [...]

Thank you for your quickly reply! It's helpful for me. On line 77 of demo4_gen_gt_files.py the frame_idx is set to 0, maybe it should be len(scan_paths)-1 according to your description

Chen-Xieyuanli, 2020-12-02:
Thanks for the suggestion!

Btw, if you want to train a rotation-invariant model, you could try to use only the delta head with rotation augmentation (random rotation or with a fixed step) and that should work. Since there is also a small CNN in the delta head, I think rotation augmentation is needed.

I'll appreciate it a lot if you could later report the results to us!

Since I currently cannot provide any further suggestions, I'll therefore close this issue and looking forward to your results.

Please feel free to reopen this issue if needed.

laebe, 2020-12-07:
> 
> Thank you for your quickly reply! It's helpful for me. On line 77 of demo4_gen_gt_files.py the frame_idx is set to 0, maybe it should be len(scan_paths)-1 according to your description

I looked again at the code and checked it: For the training we are usually looking backward, thus compare the last frame versus the previous ones. As this is a demo code and we wanted a visualization (overlap one frame versus all others) we used frame 0 because it is easy to describe and gives a nice figure because this is the overlapping area of the trajectory. For real training you will need more pairs. I guess everything will also work if you look forward, but looking back is more natural for loop closing. 

But still your question was very useful, because I found a bug in com_overlap_yaw.py: The index of the currrent frame was wrong in the return value ! But this does not influence the visualization.

Thanks again and feel free to further comment and ask questions!

closed 2020-12-07 by laebe