PRBonn/OverlapNet #13: Reproducing results of Fig.5 in paper

Daideyun, 2020-12-14:
Hi!
Thanks for your awesome work!
I have some questions. When you use RangeNet + + to get semantic clues, Are the parameters of RangeNet++ same between KITTI sequence and Ford campus sequence? 
If It is different, how do I adjust parameters? 
Do I need to train the RangeNet++ by using Ford campus dataset?

Chen-Xieyuanli, 2020-12-14:
Hey @Daideyun,

Thanks for following our work!

We didn't use the semantic information when testing on the Ford campus sequence and we found without semantic information our method also works well. 

Since the environment and the sensor setup are different between the KITTI and Ford campus dataset, you may need to fine-tune the model a little bit.

There were also other users who tried to use RangeNet++ on the ford campus dataset, where you may find more information: https://github.com/PRBonn/rangenet_lib/issues/28

I hope this helps!

Daideyu, 2020-12-16:
Ok!
Thank you very much! @Chen-Xieyuanli

Chen-Xieyuanli, 2020-12-16:
Thanks for the reply!

I'm about to close this issue. If there are further questions regarding this, please feel free to ask me to reopen it.

closed 2020-12-16 by Chen-Xieyuanli