PRBonn/OverlapNet #14: Questions about my test

hitbg-yjm, 2020-12-31:
Hi, I test OverlapNet in other datasets and the results are shown in the picture below.Robot is in the same place with different orientation. I want to know if the orientation will affect the output. Because the reality is that robot often come back to one place with different orientation.
[image]
[image]

Chen-Xieyuanli, 2021-01-01:
Hey @hitbg-yjm,

Happy New Year! Thanks for following our work!

That's a good question. Theoretically, the overlap estimation should be invariant to the rotation as can be seen from the eq3 in the paper. However, since we trained the two heads together, the correlation head will also affect the weights of the legs too.

There is a similar issue #12 discussing the rotation invariant problem which you may find useful. It's also about how to make the overlap prediction less influenced by the orientation which is actually to use the rotation augmentation during the training process.

I hope my answer helps!

hitbg-yjm, 2021-01-04:
Thanks for your answer! And Happy New Year!

closed 2021-01-04 by hitbg-yjm