PRBonn/OverlapNet #15: Questions about overlapnet

Miraclelzk, 2021-01-05:
This is a great study
I want to compare my method with yours in other dataset. 
1.   But I don't want to train model by myself. Can I use your trained model to compare directly？
2.  The input mltiple cues has 4 channels, can I only use the first three channels for verification, because obtaining semantic information is complicated for me.

Thank you and great work!

Chen-Xieyuanli, 2021-01-06:
Hey @Miraclelzk,

Thank you for following our work! You could find the pre-trained model here: http://www.ipb.uni-bonn.de/html/projects/overlap_net/model_geo.weight.

We currently only provide the pre-trained model with depth and normal which doesn't include semantic information. 

I hope my answer helps.

Chen-Xieyuanli, 2021-05-02:
Hey @Miraclelzk,

Since there is no further update, I would like to close this issue. 

If you have any further questions please feel free to ask me to reopen it!

closed 2021-05-02 by Chen-Xieyuanli