PRBonn/OverlapNet #17: Question about covariance ?

bnblzq, 2021-02-10:
Dear Sir,
   You have mentioned the covariance data are included into the odometry folder,  And I wonder how can you get the covariance from KITTI dataset ? Or where you suggested the place I can get the reference ?

Thank you for your attention~

Chen-Xieyuanli, 2021-02-16:
Hey @bnblzq,

Sorry for the late reply. 

The covariance estimation is not a part of OverlapNet but from the SLAM pipeline.
For more information about all mathematic details, you could find it here in Peter J. Huber's Robust statistics.

So far, we don't have a plan to combine the SLAM repo with this repo, and cannot provide any further help.
I will therefore close this issue.

closed 2021-02-16 by Chen-Xieyuanli

where2go947, 2024-03-04:
Dear Author,

Thank you for sharing your work. I found you are also the author of the great work [SuMa++](https://github.com/PRBonn/semantic_suma/tree/99df4940eb6b1d5a3fcb463b33a843514d48bfda). 

I'd like to get the covariance matrix from the KITTI dataset (other seqs maybe). However, the entire code of SuMa++ is a bit difficult for me to understand, and I am not able to modify it well to make my purpose work.

All I need is to generate the covariance result. Could you please give me some help on it?

Looking forward, thanks!

https://github.com/PRBonn/semantic_suma/blob/99df4940eb6b1d5a3fcb463b33a843514d48bfda/src/rv/Math.cpp#L27-L43