PRBonn/OverlapNet #18: Question about computing overlap ground truth

chienerh, 2021-02-15:
Hi, thanks for your great work!

I have a question while looking into how you generate the ground truth (demo 4).
In the paper, you mentioned that you use P_1 and P_2 and compute the overlap value with equation 3. And the result of it is figure 2 (a).
However, I am still not quite sure how you compute the overlap value ground truth shown in figure (c). 
By looking into the [com_overlap_yaw() function](https://github.com/PRBonn/OverlapNet/blob/c51878be163143cfe9b5425fc0d1b888e9d1d163/src/utils/com_overlap_yaw.py#L43), you seem to compare the same point cloud with a different pose.
What confuses me is that reference_range and current_range are using the same point cloud, not P_1 and P_2.

It would be great if you could explain more. Thanks!

Chen-Xieyuanli, 2021-02-16:
Hey @chienerh,

Thank you for the feedback. I will carefully review the code and reply to you soon.

Chen-Xieyuanli, 2021-02-17:
Hey @chienerh,

Thank you so much to point the problem out! 

You are right. The old script is wrong, which may be caused when we created the demo.

I've already corrected the function and please have a check [here](https://github.com/PRBonn/OverlapNet/blob/master/src/utils/com_overlap_yaw.py)

Thank you again for following our work and help us to improve our code!

Best,
Xieyuanli

chienerh, 2021-02-18:
Hi @Chen-Xieyuanli,

Thanks a lot for the quick responding. 
The updated code makes sense for me now. 
Thanks for your update!

closed 2021-02-18 by chienerh

yiming107, 2021-09-14:
Hi @Chen-Xieyuanli, following this issue, I see the com_overlap_yaw() function, when you compute the overlap:
`overlap = np.count_nonzero(
          abs(reference_range[reference_range > 0] -  current_range[reference_range > 0]) < 1) / valid_num
        overlaps.append(overlap)`
I understood the idea is to account only valid points, so I assume, we should have **current_range[current_range > 0]** rather than **current_range[reference_range > 0]** here? Let me know what do you think, thanks.