PRBonn/OverlapNet #20: Preprocessing time

cattaneod, 2021-06-14:
Dear Authors,

in your paper the time required to generate the geometric input features is reported to be around 10ms.
However, with this code on my computer it takes more than 2 seconds to generate the normal image.

Did you use a different implementation to achieve the runtime reported in your paper? Or am I doing something wrong?

Chen-Xieyuanli, 2021-06-14:
Hey @cattaneod,

Thanks for using our code.

As mentioned in the readme, for the fast LiDAR scan preprocessing you could use our [SuMa++](https://github.com/PRBonn/semantic_suma) which exploits OpenGL shaders to fast rendering the range and normal images.
We use that for the online SLAM and loop closing, where the preprocessing of the scan takes around 10 ms.

For generating range and normal training data without using GPU and OpenGL, we also provide a fast C++ implementation [here](https://github.com/PRBonn/overlap_localization/tree/master/src/prepare_training), which is faster than the python implementation.

I hope this helps.

cattaneod, 2021-06-15:
Thank you @Chen-Xieyuanli,

I tried the C++ implementation, and it's much faster than the python version, it takes around 9ms for the normal computation, so similar to the time reported in the paper.

closed 2021-06-15 by cattaneod