PRBonn/OverlapNet #22: Question about how to judge a loop closure True Positive or not

lymdove, 2021-08-30:
Hi, thanks for your great work!

I have a question about how to judge a loop closure True Positive or not. Some papers use the distance between two frames, think it True Positive if the distance less than 3 meters or 4 meters. Is OverlapNet use this method or think it true if overlap lager than threshold.

It would be great if you could explain more. Thanks!

Chen-Xieyuanli, 2021-08-30:
Hey @lymdove, thanks for following our work. Yes, I use the overlap threshold to determine the ground-truth loop closures.
You could find more information in the [paper](https://www.ipb.uni-bonn.de/wp-content/papercite-data/pdf/chen2020rss.pdf).
"We use a relatively low threshold of 30 % for the overlap to decide if a candidate is a true positive. We aim to find more loops even in some challenging situations with low overlaps, e.g., when the car drives back to an intersection from the opposite direction (as highlighted in the supplementary video1). Furthermore, ICP can find correct poses if the overlap between pairs of scans is around 30 %, as illustrated in the experimental evaluation."

lymdove, 2021-08-30:
@Chen-Xieyuanli Thanks for your replying.
I also major in a new method to detect loop closure in some challenging situations. But i met some problems, like the judgement in your work, some loops may have low overlap but ICP can calculate correct poses(the loop will make SLAM better). If use overlap threshold to judge the true positive, loops mentioned above would be thought negetive, this situation will reduce the recall. I am not sure i should propose a new judgement or do some other things. Could you give me some advices?

Chen-Xieyuanli, 2021-08-30:
That's also the reason why I use overlap as the threshold since we want to find as many loops as possible.
That's also one of the contributions of this work to find a better way to distinguish loops.
I would suggest still use overlap or other ways of similarity estimation to decide the ground truth rather than the distances.

lymdove, 2021-08-30:
Thanks for axplaination.

closed 2021-08-30 by lymdove