PRBonn/OverlapNet #25: some errors encountered during training

shenhai911, 2021-12-19:
Dear author,
when I try to train the model, I always encounter the following error:

`
Traceback (most recent call last):
  File "src/two_heads/training.py", line 351, in <module>
    model.save(weights_filename)
  File "XXX/anaconda3/envs/OverlapNet_env/lib/python3.7/site-packages/keras/engine/topology.py", line 2580, in save
    save_model(self, filepath, overwrite, include_optimizer)
  File "XXX/anaconda3/envs/OverlapNet_env/lib/python3.7/site-packages/keras/models.py", line 111, in save_model
    'config': model.get_config()
  File "XXX/anaconda3/envs/OverlapNet_env/lib/python3.7/site-packages/keras/engine/topology.py", line 2353, in get_config
    layer_config = layer.get_config()
  File "XXX/anaconda3/envs/OverlapNet_env/lib/python3.7/site-packages/keras/layers/convolutional.py", line 471, in get_config
    config = super(Conv2D, self).get_config()
  File "XXX/anaconda3/envs/OverlapNet_env/lib/python3.7/site-packages/keras/layers/convolutional.py", line 231, in get_config
    'bias_initializer': initializers.serialize(self.bias_initializer),
  File "XXX/anaconda3/envs/OverlapNet_env/lib/python3.7/site-packages/keras/initializers/__init__.py", line 132, in serialize
    return generic_utils.serialize_keras_object(initializer)
  File "XXX/anaconda3/envs/OverlapNet_env/lib/python3.7/site-packages/keras/utils/generic_utils.py", line 131, in serialize_keras_object
    'config': instance.get_config()
TypeError: get_config() missing 1 required positional argument: 'self'
`

Although I have tried various solutions to solve this problem, I still can't solve it. Thank you very much for your advice.
Best wishes to you.

Chen-Xieyuanli, 2021-12-19:
Hey @shenhai911, thanks for using our code.

It seems to be a TensorFlow version conflict. I never met such a problem and just googled around.
Here I found some similar issues in the StackOverflow which you may it helpful [link](https://stackoverflow.com/questions/57154799/keras-model-saving-erroring-typeerror-get-config-missing-1-required-position).

shenhai911, 2021-12-19:
> Hey @shenhai911, thanks for using our code.
>
> It seems to be a TensorFlow version conflict. I never met such a problem and just googled around. Here I found some similar issues in the StackOverflow [...]

Thank you very much for your quick reply. I have tried these methods, but they didn't work.
I think it may be because of the version of the tensorflow I used, but I installed it according to the latest **_requirements.txt_**.
I will try again.
Thank you again and best wishes to you.

shenhai911, 2021-12-21:
Dear author, I'm so sorry that I still can't solve the above problem. I would like to ask, what's the version of Python you are using?
thank you very much for you reply.
Best wishes to you.

shenhai911, 2021-12-21:
Dear author, the hardware and software configuration of my computer is as follows:
                      `RTX3070，ubuntu20.04，anaconda, python3.7.11`

but I encountered a series of problems related to Tensorflow during test, many of which were keras pointing problems.
By the way, I am a beginner in deep learning, thus I'm not familiar with tensorflow, so some questions may be rather naive, please forgive my ignorance.
Thank you very much for your guidance. 
Best wishes to you.

Chen-Xieyuanli, 2022-01-04:
Hey @shenhai911, sorry for the late reply due to the holidays.

We developed OverlapNet with python3.6 and TensorFlow 1.14.0.
Junyi Ma recently provided a PyTorch version. If you still have problem with the newer TensorFlow, you may also try the PyTorch version here: https://github.com/PRBonn/OverlapNet/tree/pytorch-version

shenhai911, 2022-01-05:
> Hey @shenhai911, sorry for the late reply due to the holidays.
>
> We developed OverlapNet with python3.6 and TensorFlow 1.14.0. Junyi Ma recently provided a PyTorch version. If you still have problem [...]

Dear author,
Thank you very much for your guidance. The problems encountered before are mainly because of the mismatching versions of some dependent libraries, which have been solved at present. Thank you again for your selfless work.

Chen-Xieyuanli, 2022-01-05:
> 

Thank you @shenhai911 for using our code and for your nice feedback!

I would like to close this issue and feel free to reopen it if there are still some other problems.

closed 2022-01-05 by Chen-Xieyuanli