PRBonn/OverlapNet #29: Requirements of pytorch

smalltheater, 2022-03-17:
Hello, what's the requirements of pytorch version?

Chen-Xieyuanli, 2022-03-17:
@BIT-MJY could you please give your PyTorch setup here?

BIT-MJY, 2022-03-18:
Hello @smalltheater ,

Thanks for using our code! The setup of OverlapNet (pytorch) is the same as the one used in our newest work [OverlapTransformer](https://github.com/haomo-ai/OverlapTransformer), which is:  
```
torch==1.10.2+cu113  
torchvision==0.11.3+cu113  
torchaudio==0.10.2+cu113  
  
matplotlib==3.4.3    
numpy==1.20.3    
tensorboardx==2.4    
pyyaml==6.0    
opencv-python==4.5.4.58    
faiss-cpu==1.7.1    
scikit-learn==0.24.2    
```

You can download our Torch dependences by pip:  

`pip3 install torch==1.10.2+cu113 torchvision==0.11.3+cu113 torchaudio==0.10.2+cu113 -f https://download.pytorch.org/whl/cu113/torch_stable.html`

or by conda:  
`conda install pytorch torchvision torchaudio cudatoolkit=11.3 -c pytorch
`

Chen-Xieyuanli, 2022-03-21:
@BIT-MJY Thanks for the detailed reply.