PRBonn/OverlapNet #30: Some confusion about real overlap calculation

ZYJ0409, 2022-04-11:
Dear author
    I had some problems running your code with the overlap truth calculation. In the 00 sequence, frame 0 and frame 4449 are the same region, but their true overlap is only 0.14.
[image]
[image]
I look forward to your reply. Thank you

Chen-Xieyuanli, 2022-04-11:
Hey @ZYJ0409, thanks for using our code!

In my case, it's more than 80. Which code are you using? You may double check the poses you are using.

[image]

ZYJ0409, 2022-04-11:
Dear author
Thank you for your reply. I refer to '**com_overlap_yaw ()**' in your code. I enter '**current_POSES**' in frame 0 (white) and '**reference_IDx**' in frame 4449 (blue). This is the pose of my two frames, but the overlap area of my output is only 0.14. How do you solve the error caused by the scale drift of kitti dataset?
[image]
[image]
[image]
[image]
Looking forward to your reply.

Chen-Xieyuanli, 2022-04-11:
I think the error comes from the poses. Are you using the ground truth poses? They are in the camera coordinate systems and you need to first convert them into the LiDAR coordinate system.

ZYJ0409, 2022-04-12:
Dear author
I'm sorry that's my problem. I didn't express myself clearly. This is indeed the camera pose, but your code has been converted in "demo4_gen_gt_files", I have taken this into account, but the result is still very low
[image]

Chen-Xieyuanli, 2022-04-12:
How did you generate the poses? They seem not accurate and not the ground truth poses.

ZYJ0409, 2022-04-12:
Dear author
    I downloaded them from the Internet， Can you provide a link to download the ground truth poses? (the best is '**baidu cloud**’)，thank you very much, Your reply was very helpful to us!

ZYJ0409, 2022-04-12:
Thank you very much for your patient help. There is something wrong with our data. Wish you a happy life and smooth work!

Chen-Xieyuanli, 2022-04-12:
I'm glad that the problem is solved! I will then close this issue.

I'll appreciate it if you could star our repo to let more people see it if you find our code and answers helpful.

closed 2022-04-12 by Chen-Xieyuanli

BlingLee, 2024-08-28:
[image]
> 非常感谢您的耐心帮助。我们的数据有问题。祝您生活愉快，工作顺利！
Hello. How are you? I also tried to reproduce overlapnet, but I met some problems, can I ask you for advice?

In demo4, we need to enter poses.txt and calib.txt as well as scan. May I ask what processing is required for this poses. I download the poses file for the kitti dataset directly from the Internet, and the authors of the paper also provide the poses file for the 07 sequence, but for the 07 sequence, the file that I download is not the same as the Poses file provided by the authors. Okay? poses.txt is converted in demo4.
[image]

This is the 07 sequence poses.txt that I downloaded. Looking forward to your reply