PRBonn/OverlapNet #33: Pytorch version training.

lymdove, 2022-09-13:
Thanks for your great work!
I want to compare my lopp closure method with Overlap, but I have problem with running the work. I use the pytorch version, and I already generate depth, intensity, normal data. I want to compute a score for every pair, but I have no model weights. So I want to train one, but I can not find which file could generate the `'overlaps/train_set.npz'`. Could you help me, or could you provide a pre-model. Thanks.

Chen-Xieyuanli, 2022-09-22:
@BIT-MJY could you please provide the pre-trained model using PyTorch?

@lymdove you may also have a look at our novel overlaptransformer: https://github.com/haomo-ai/OverlapTransformer
It is implemented using PyTorch, and the pre-trained model was given.

surajiitd, 2022-11-02:
@BIT-MJY  could you please provide the pre-trained model using PyTorch?

surajiitd, 2022-11-05:
@lymdove `demo/demo4_gen_gt_files.py` can generate the `train_set.npz` file in `data/{seq_num}/ground_truth/train_set.npz`

BlingLee, 2024-07-14:
> 感谢您的出色工作！我想将我的 lopp 关闭方法与 Overlap 进行比较，但我在运行工作时遇到了问题。我使用 pytorch 版本，我已经生成了深度、强度、正常数据。我想计算每对的分数，但我没有模型权重。所以我想训练一个，但我找不到哪个文件可以生成.你能帮我吗，或者你能提供一个预模型吗？谢谢。`'overlaps/train_set.npz'`

hi，How do you use the pytorch version to generate information like depth and strength? Looking forward to your reply very much！

laebe, 2024-07-16:
I'm not sure, because I was not involved in the pytorch version. But my guess is that the pytorch version uses the input which can be generated by the non-pytorch version (as stated above e.g. with demo4)