PRBonn/OverlapNet #34: Questions when Generating train_set

Bazinga92445, 2022-10-26:
Dear Author,
Thank you for sharing your work, and I enjoy your paper very much.
I found some problems when I tried to generate train_set under ground_truth folder:
1. The length of the array "train_set" (which is the result I need) is always shortened greatly after the process of normalization, showing that the number of overlaps between 0.4 and 0.5 is quite small. If my understanding of your code is right, the training set will be extremely small (which isn't).  For example, sequence 05 has 2761 frames, but in the process, the size of normalized data(printed by function normalize_data) is only 31.
[image]
2. I downloaded your provided 'train_set' files when viewing the github page of OverlapTransformer (which claims that: "overlaps folder of each sequence below data_root_folder is provided by the authors of OverlapNet [here](https://drive.google.com/file/d/1i333NUC1DnJglXasqkGYCmo9p45Fx28-/view?usp=sharing). ").  However, I found that the files are extremely large causing slow training speed. For example, sequence 03 has only 801 frames. In theory, since demo4 only implement the function "com_overlap_yaw" once setting frame_idx = 0, the maximum length of train_set should be smaller than 801. However, when I check the length of the array, it appears to be an astonishing number of 16218. I don't understand where did that come from. I guess I've missed something, but I can't figure out what it is.
[image]
[image]
Looking forward to your reply!

BIT-MJY, 2022-10-27:
Hi @Bazinga92445 Thanks for using our code!

Demo 4 only uses the first frame of the sequence to generate the gt mapping by   
`com_overlap_yaw(scan_paths, poses, frame_idx=0)`  

Actually to generate the train_set.npz provided by OverlapTransformer, we need to consider all the laser scans rather than only the first one. In theory, the total number of the mapping pairs is NxN (e.g. 801X801 for seq. 3) which is very large since overlap can be calculated between every anchor-reference laser pair. Therefore, normalization is used here to guarantee the reasonable distribution of the overlap-labeled pairs and reduce the amount of data.

To achieve this you need to extend the code of Demo 4 in this repo to generate overlap and yaw ground truth array for each laser scan of the sequence, e,g,

```
for n in range(N): 
    ground_truth_mapping_n = com_overlap_yaw(scan_paths, poses, frame_idx=n)
```

closed 2022-10-31 by Bazinga92445

surajiitd, 2022-11-05:
@Chen-Xieyuanli @laebe 
> To achieve this you need to extend the code of Demo 4 in this repo to generate overlap and yaw ground truth array for each laser scan of the sequence, e.g.

@BIT-MJY Have you done this for OverlapNet also? 
or just done for frame_idx=0 ??
When I generated overlap and yaw ground truth array for each laser scan of the sequence, **then it is going to take 5,6 days to train... Is it correct??**

laebe, 2022-11-08:
A training time of 5-6 days sound a bit long long to me, but this of course depends on your computational resources. Have you used the normalization from normalize_data.py ? If you calculate the overlap between all pairs of scans, most of the pairs have no or very very little overlap. So this results in a distribution where you have far too many low overlap pairs in your training set. 

So we used the normalize_data function to reduce the number of low overlap pairs. (The cut in this function 
is a heuristic which uses histogram bins and maybe can be improved for other data)

surajiitd, 2022-11-08:
Yes I have also used normalize_data.py file.
Before normalization, 
**Kiiti seq 00:**
no. of scans = 4541.
So I got = n*(n+1)/2 = 4541*4542 / 2 <kbd>comparing all scans with all previous scans</kbd>
= 10,312,611 queries in `ground_truth_overlap_yaw.npz`
And after Normalization: 
I got 1,52,365 queries in `train_set.npz` and 
16,929 queries in `validation_set.npz` <kbd>which are pretty low compared to original</kbd>
Is it correct??

Few Queries:
1. bin_50_59 to bin -80_89 are not sampled like others... Is it commented intentionally?
2. BTW I am using your **pytorch code**. and the loss is converging(see the pic below). but it is taking too much time to train just one epoch is taking **91min** (total 100 epochs are required acc. to your code). I have access to 8 gpus buth the code can use only single gpu as the dataloader code is not there.
3. Even though the loss is converging, the model(used intermediate saved model after 38 epochs) is **predicting overlap only in b/w 0.49 to 0.51 for all the pairs of scans**. Please help what could be the issue here??

[image]

BlingLee, 2024-09-21:
> 非常感谢您的回复！！是的，我也用过 file。 _归一化前：_示例：对于 **Kiiti seq 00：**否。扫描次数 = 4541。 所以我得到 = n*（n+1）/2 = 4541*4542 / 2 = 10,312,611 个查询 通过查询 我的意思是从提取的 .npz 文件中重叠数组的形状将是 （10312611 ， 4）。`normalize_data.py`(comparing all scans with all previous scans)`ground_truth_overlap_yaw.npz`
> 
> _规范化后：_我收到了 1,52,365 个查询和 16,929 个查询正确吗？？`train_set.npz``validation_set.npz`which are pretty low compared to original
> 
> 查询：
> 
> 1. bin_50_59 到 bin_80_89 不像其他  那样进行采样......[是故意评论](https://github.com/PRBonn/OverlapNet/blob/4188c3a22180ea6e34f5022a629971b49c60eefb/src/utils/normalize_data.py#L39)的吗？
> 2. 顺便说一句，我正在使用您的 **pytorch 代码**。并且损失正在收敛（请看下面的图片）。但是训练时间太长了，仅一个 epoch 就需要 **91 分钟**（根据您的代码总共需要 100 个 epoch）。我可以访问 8 个 40GB GPU，但 pytorch 代码只能使用单个 GPU（仅使用 3GB），因为数据加载器代码不存在。
> 3. **[小鬼]**即使损失正在收敛，模型（在 38 个时期后使用中间保存的模型）**也仅预测所有扫描对的黑白重叠为 0.49 到 0.51......大多数是 0.5**。（它应该是 b/w 0 到 1）请帮忙这里可能有什么问题？？
> 4. 顺便说一句：在 demo4 中，您刚刚将 frame_idx=0 与所有扫描 （=4541） 进行了比较。当我只使用这些数据进行训练时，训练时间更少（100 个时期 1 小时）。但是当我将所有扫描与他们之前的扫描进行比较时（您在一个问题中提到），那么仅在 1epoch 中就需要 1.5 小时）。当我将所有扫描结果与所有其他扫描结果进行比较时，会花费更多时间。  [image]

Hello, have you solved the problem yet? I have the same problem as you. When I trained with all the clues, it would take me about two hours to train about 10 epochs a day for one epoch. For demo4_gen_gt_files.py, I calculate overlap for each liDAR scan sequence, and the 00 sequence alone takes two or three days. Is this normal? Looking forward to your reply!

BlingLee, 2024-09-21:
> @laebe Thanks a lot for your reply !! Yes I have also used `normalize_data.py` file. _Before normalization:_ Example: For **Kiiti seq 00:** no. of scans = 4541. (comparing all scans with all previous scans) So I got = n*(n+1)/2 = 4541*4542 / 2 = 10,312,611 queries in `ground_truth_overlap_yaw.npz` By queries I mean the shape of the overlaps array from extracted .npz file would be (10312611 , 4).
> 
> _And after Normalization:_ I got 1,52,365 queries in `train_set.npz` and 16,929 queries in `validation_set.npz` which are pretty low compared to original Is it correct??
> 
> Queries:
> 
> 1. bin_50_59 to bin_80_89 are not sampled like others... [Is it commented intentionally](https://github.com/PRBonn/OverlapNet/blob/4188c3a22180ea6e34f5022a629971b49c60eefb/src/utils/normalize_data.py#L39)?
> 2. BTW I am using your **pytorch code**. and the loss is converging(please see the pic below). but it is taking too much time to train, just one epoch is taking **91min** (total 100 epochs are required acc. to your code). I have access to 8 x 40GB GPUs but the pytorch code can use only single gpu(using 3GB only) as the dataloader code is not there.
> 3. **[imp]** Even though the loss is converging, the model(used intermediate saved model after 38 epochs) is **predicting overlap only in b/w 0.49 to 0.51 for ALL the pairs of scans...mostly are 0.5**. (It should be b/w 0 to 1)
>    Please help what could be the issue here??
> 4. BTW In the demo4 you have just compared frame_idx=0 with all the scans(=4541). When I trained using only that data, it took v. less to train(1hr for 100epochs). but when I compared all the scans with their previous scans(you mentioned in one issue), then it is taking 1.5hr just in 1epoch). And when I compared all scans with all the other scans, then it is taking even more time.
>    [image]
>    @BIT-MJY @Chen-Xieyuanli

Hello, have you solved the problem yet? I have the same problem as you. When I trained with all the clues, it would take me about 10 epochs a day.  For one epoch,two hours to train . For demo4_gen_gt_files.py, I calculate overlap for each liDAR scan sequence, and the 00 sequence alone takes two or three days. Is this normal? Looking forward to your reply!