PRBonn/OverlapNet #35: Pytorch version training

surajiitd, 2022-10-27:
In your [recommended Data Structure](https://github.com/PRBonn/OverlapNet/tree/pytorch-version#data-structure) how to generate `loop_gt_seq00_0.3overlap_inactive.npz` file for training? 
I have used the tensorflow branch to generate the preprocessed data, but no such file is generated.

BIT-MJY, 2022-10-27:
Hi @surajiitd Thanks for using our code!

loop_gt_seq00_0.3overlap_inactive.npz contains _N_ lists for _N_ laser scans of sequence 00. In each list _n_, there are indices of laser scans that have overlap > 0.3 with respect to the _n_th laser scan (exclude near 100 scans). As to the calculation of overlap, you can refer to OverlapNet [Demo4](https://github.com/PRBonn/OverlapNet/blob/master/demo/demo4_gen_gt_files.py).

BTW, you can find our newer works OverlapTransformer and SeqOT to implement faster retrieval. We have released the code to generate distance-based loop git files for NCLT dataset [here](https://github.com/BIT-MJY/SeqOT/blob/main/data_prepararion/gen_ground_truth.py). The data structures are quite similar. The only difference is that each list in git files for NCLT contains the loop indices in database.

surajiitd, 2022-10-29:
@BIT-MJY Thank you so much for your instant reply!
But I already ran [Demo4](https://github.com/PRBonn/OverlapNet/blob/master/demo/demo4_gen_gt_files.py) of OverlapNet and tried again. But I can't generate any `loop_gt_seq00_0.3overlap_inactive.npz` file. No such file is generated that has such info (one list for each laser scan). Please help on how to get this file!
Of course, I'd like to see your more recent work [OverlapTransformer](https://github.com/haomo-ai/OverlapTransformer) and planning to use it. But that is also [referring](https://github.com/haomo-ai/OverlapTransformer#testing) to Overlapnet for generating `loop_gt_seq00_0.3overlap_inactive.npz` file.

BIT-MJY, 2022-10-30:
We have uploaded our newest version of this related script to generate ground truth files in the OT repo. You can find it [here](https://github.com/haomo-ai/OverlapTransformer/blob/master/tools/utils/gen_gt_data.py).

Please pull our new code, modify the paths in this file (`tools/utils/gen_gt_data.py`), and run

```
cd tools/utils/
python gen_gt_data.py
```

Note that the poses file is from [semantic kitti](http://www.semantic-kitti.org/dataset.html#download), which is refined by SLAM.

BIT-MJY, 2022-11-01:
I suppose so, but we recommend using semanticKitti poses in your work which are more accurate.

surajiitd, 2022-11-01:
Thank you so much @BIT-MJY

closed 2022-11-01 by surajiitd

LY0406, 2024-07-04:
您好，我正在学习您的OverlapTransformer这篇论文，您在论文中给出的新数据集Haomo数据集似乎仅仅给出了激光序列和位姿文件而没有提供训练OT所需要的训练集，如果我想要得到训练集应该怎么做呢。

laebe, 2024-07-04:
[LY0406](https://github.com/LY0406), would be nice to write in english so that everybody (include me) could read it !