PRBonn/OverlapNet #36: Problem in the first version

ydragon719, 2022-11-03:
I used Ubuntu18.04 to run it in terminl. But when i run the demo2_infer.py , i meet the following problem 

2022-11-03 20:27:14.318140: I tensorflow/stream_executor/platform/default/dso_loader.cc:53] Successfully opened dynamic library libcudart.so.11.0
Traceback (most recent call last):
  File "/home/ydragon/Downloads/OverlapNet-master/demo/demo2_infer.py", line 15, in <module>
    from infer import *
  File "/home/ydragon/Downloads/OverlapNet-master/demo/../src/two_heads/infer.py", line 11, in <module>
    import keras
  File "/home/ydragon/anaconda3/envs/tensorflow/lib/python3.9/site-packages/keras/__init__.py", line 3, in <module>
    from . import utils
  File "/home/ydragon/anaconda3/envs/tensorflow/lib/python3.9/site-packages/keras/utils/__init__.py", line 25, in <module>
    from .multi_gpu_utils import multi_gpu_model
  File "/home/ydragon/anaconda3/envs/tensorflow/lib/python3.9/site-packages/keras/utils/multi_gpu_utils.py", line 7, in <module>
    from ..layers.merge import concatenate
  File "/home/ydragon/anaconda3/envs/tensorflow/lib/python3.9/site-packages/keras/layers/__init__.py", line 4, in <module>
    from ..engine import Layer
  File "/home/ydragon/anaconda3/envs/tensorflow/lib/python3.9/site-packages/keras/engine/__init__.py", line 3, in <module>
    from .topology import InputSpec
  File "/home/ydragon/anaconda3/envs/tensorflow/lib/python3.9/site-packages/keras/engine/topology.py", line 18, in <module>
    from .. import initializers
  File "/home/ydragon/anaconda3/envs/tensorflow/lib/python3.9/site-packages/keras/initializers/__init__.py", line 124, in <module>
    populate_deserializable_objects()
  File "/home/ydragon/anaconda3/envs/tensorflow/lib/python3.9/site-packages/keras/initializers/__init__.py", line 82, in populate_deserializable_objects
    generic_utils.populate_dict_with_module_objects(
AttributeError: module 'keras.utils.generic_utils' has no attribute 'populate_dict_with_module_objects'

Strangely ,I didn't enconter this problem in demo1_gen_data.py with"from utils import *"  and "  import keras". what's wrong with it in demo2_infer.py and infer.py? Could you help me solve it ? Thank you very much .

laebe, 2022-11-08:
Thanks [ydragon719] for you interest in our work !

It could be that this is a keras/tensorflow problem (but I'm not sure). I have found this:
https://github.com/keras-team/keras/issues/14632

ydragon719, 2023-04-23:
Thanks you! I have finished it. But I have a problem for the ground truth  of the loop .  How do you get the ground truth of the loop close decetion  in kitti? The code in the project should be to obtain the true value of the network.

closed 2023-04-24 by laebe